RPi-Monitor/RPi-Monitor #71: unable to install the new version 2.10

anddag, 2015-01-01:
I've followed the installation instruction but "apt-get install " install the old version..

XavierBerger, 2015-01-01:
Oups, the version 2.10 is in fact not yet officially published. A mistake in my blog made the article presenting v2.10 been published a little bit in advance.
I will soon generate a release candidate of the debian package in the devel branch.
Help for tests will then be welcome.
I'll update this post when it will be ready.

anddag, 2015-01-01:
ok,Can I be a tester?
I'm playing with the raspi (i've configured a Raspi witch, cups, saned,
samba) ed raspi monitor as web interface
ciao
andrea

XavierBerger, 2015-01-02:
I just pushed `rpimonitor_2.10-1_all.deb` in devel branch (in RPi-Monitor-deb repository), directory packages. It can be downloaded from github and installed with the command `dpkg -i rpimonitor_2.10-1_all.deb`.
You can add your feedbacks by commenting this issue.
Thanks for your help.

anddag, 2015-01-02:
i've installed this .deb but i see the error:
Can't use an undefined value as an ARRAY reference at /usr/bin/rpimonitord
line 498, <FILE> line 370.
Can't use an undefined value as an ARRAY reference at /usr/bin/rpimonitord
line 498, <FILE> line 370.

XavierBerger, 2015-01-03:
The issue listed upper is now fixed. 
I also add in devel branch of RPi-Monitor-deb repository, the article I plan to publish explaining the new feature aded in version 2.10 (raw html from blogger...). This may help you in your tests.

anddag, 2015-01-03:
I've upgraded the package. Running as daemon it works.
Running with option
daemon.noserver=1
i got the error
Not a HASH reference at /usr/bin/rpimonitord line 601.
rpimonitord stopped because respawning too fast.

In the .conf I use the user root and the group root. The file are copied in
www of nginx.

XavierBerger, 2015-01-03:
I was conncted on my PC when you log this issue... A quick look show me a '%' instead a '@'  in line 601. Fix is done.

anddag, 2015-01-03:
now rpimontord is running but the status page o statistics is white....
I've copied all files in webroot of ngnix e modified the .conf to save the
.json file there.

anddag, 2015-01-03:
there is a problem in the creation of menu.json
the createad file is only two bytes:

{}

2015-01-03 19:15 GMT+01:00 andrea d'agostino <anddag@gmail.com>:

XavierBerger, 2015-01-04:
Problem is fixed. It was a side effect of new 'addons' feature.
I did some test with python web server (`python -m SimpleHTTPServer`) and it works for me.

XavierBerger, 2015-01-04:
Note: I just update v2.10 package now to fix issue found during addons tests and validation.

anddag, 2015-01-04:
now is ok.. I've cofigure top3 and custom addons .
In the custom addon I believe that there is a bad use of
window.localstorage (this must only used to have a permanent persistence).

XavierBerger, 2015-01-06:
Version 2.10 published

closed 2015-01-06 by XavierBerger