RPi-Monitor/RPi-Monitor #72: question about the id's

maphy-psd, 2015-01-03:
Hi!

First of all..RPi-Monitor is a really nice program. I love it.

But i have a question about the id's. Every status strip has his own configuration file. In the raspbian.conf are all included. I found on your website, that every single data source and status strip must be have his own id. But the RPi-Monitor works without that as well.

For example the raspbian.conf contains:
```
include=/etc/rpimonitor/template/version.conf
include=/etc/rpimonitor/template/memory.conf
```
and the version.conf
```
static.1.name=distribution
static.1.source=/etc/os-release
static.1.regexp=PRETTY_NAME.\"(.*)\"
static.1.postprocess=

static.2.name=kernel_version
static.2.source=uname -msr
static.2.regexp=(.*)
static.2.postprocess=

static.3.name=firmware
static.3.source=/proc/version
static.3.regexp=(#\d+)
static.3.postprocess=

static.4.name=processor
static.4.source=/proc/cpuinfo
static.4.regexp=(?:Processor|model name)\s+: (.*)
static.4.postprocess=

dynamic.1.name=upgrade
dynamic.1.source=/var/lib/rpimonitor/updatestatus.txt
dynamic.1.regexp=(\d+ upgradable\(s\)|.*&nbsp;.*)
dynamic.1.postprocess=
dynamic.1.rrd=

dynamic.2.name=packages
dynamic.2.source=/var/lib/rpimonitor/updatestatus.txt
dynamic.2.regexp=^\s+(.*)\s+\d+ upgradable
dynamic.2.postprocess=
dynamic.2.rrd=

web.status.1.content.1.name=Version
web.status.1.content.1.icon=version.png
web.status.1.content.1.line.1='Processor: <b>' + data.processor + '</b>'
web.status.1.content.1.line.2='Distribution: <b>'+ data.distribution + '</b>'
web.status.1.content.1.line.3='Kernel version: <b>' + data.kernel_version + '</b>'
web.status.1.content.1.line.4='Firmware: <b>' + data.firmware + '</b>'
web.status.1.content.1.line.5='Package(s): <b>' + data.upgrade + '</b>' + ShowInfo('packages','Package(s) upgradable(s)',data.packages)

```
and the memory.conf
```
static.1.name=memory_total
static.1.source=/proc/meminfo
static.1.regexp=MemTotal:\s+(\d+)
static.1.postprocess=$1/1024

dynamic.1.name=memory_free
dynamic.1.source=/proc/meminfo
dynamic.1.regexp=MemFree:\s+(\d+)
dynamic.1.postprocess=$1/1024
dynamic.1.rrd=GAUGE

dynamic.2.name=memory_available
dynamic.2.source=/usr/bin/free -mk
dynamic.2.regexp=^-\/\+ buffers\/cache:\s+\d+\s+(\d+)
dynamic.2.postprocess=$1/1024
dynamic.2.rrd=GAUGE

web.status.1.content.1.name=Memory
web.status.1.content.1.icon=memory.png
web.status.1.content.1.line.1="Used: <b>" + KMG(data.memory_total-data.memory_available,'M') + "</b> (<b>" + Percent(data.memory_total-data.memory_available,data.memory_total,'M') + "</b>) Available: <b>" + KMG(data.memory_available,'M') + "</b> Total: <b>" + KMG(data.memory_total,'M') + "</b>"
web.status.1.content.1.line.2=ProgressBar(data.memory_total-data.memory_available,data.memory_total)

web.statistics.1.content.1.name=Memory
web.statistics.1.content.1.graph.1=memory_total
web.statistics.1.content.1.graph.2=memory_free
web.statistics.1.content.1.graph.3=memory_available
web.statistics.1.content.1.ds_graph_options.memory_total.label=Total Memory(MB)
web.statistics.1.content.1.ds_graph_options.memory_free.label=Free Memory (MB)
web.statistics.1.content.1.ds_graph_options.memory_free.color="#7777FF"
web.statistics.1.content.1.ds_graph_options.memory_available.label=Available Memory (MB)
web.statistics.1.content.1.ds_graph_options.memory_available.color="#77FF77"
```

So in every config file the static, dynamic and status id starts with 1. So i can change and modify the RPi-Monitor easily. My question is, is that a good way for the configuration or i get in trouble with that?

Thanks!

XavierBerger, 2015-01-03:
Yes, this is the good way to configure RPi-Monitor.

In first version of **RPi-Monitor**, it was mandatory to have overall unique `id`.
Since I introduce the `include` keyword, *`id` should be unique per file*. `id` should be an integer, whetever is its value, **RPi-Monitor** will regenerate the `id` internally to have a consistant display.

maphy-psd, 2015-01-03:
Thank you for your quick reply. I wish you a Happy New Year!

closed 2015-01-03 by maphy-psd