RPi-Monitor/RPi-Monitor #78: Network statistics reset

kropochev, 2015-01-14:
The problem is described on the link http://raspberrypi.stackexchange.com/questions/12277/network-statistics-reset

Can go to vnstat? And to add a bit rate in real time?

XavierBerger, 2015-01-18:
I didn't knew this software. Interresting tools indeed. 
We could imagine an addons displaying image generated by `vnstati` or use it directly as a source of information for **RPi-Monitor** with `vnstat --oneline` and a correctly shapped regular expression.

kropochev, 2015-01-18:
```
dynamic.10.name=net_received
dynamic.10.source=vnstat -i eth0 --dumpdb
dynamic.10.regexp=totalrx;(\d+)
dynamic.10.postprocess=$1*1024*1024*-1
dynamic.10.rrd=DERIVE
dynamic.10.max=0

dynamic.11.name=net_send
dynamic.11.source=vnstat -i eth0 --dumpdb
dynamic.11.regexp=totaltx;(\d+)
dynamic.11.postprocess=$1*1024*1024
dynamic.11.rrd=DERIVE
dynamic.11.min=0
```

Advantages over the old version, no data are reset after reboot.
Disadvantages, accuracy 1MB and the default data in the database is updated every 5 minutes, but in the config /etc/vnstat.conf you can change the update time per minute.
```
# how often (in minute) data is saved to file
SaveInterval 1
```

typhoon71, 2015-04-06:
This is indeed good for keeping net stats between reboo, but there's an issue with graphs: the values drawn are wrong.
Right now I have a 2MBps up (FTP conn), but the graph goes as high as 60MBps while using vnstat, which is not possible.
Is it possile to fix this? Tthanks.

albertoacj, 2023-02-22:
In new versions of vnstat  --dumpdb doesn't work anymore, the solution is

dynamic.1.name=net_received
dynamic.1.source=vnstat --json -i eth0 | jq '.interfaces[0].traffic.total.rx'
dynamic.1.postprocess=$1*1024*1024*-1
dynamic.1.rrd=DERIVE
dynamic.1.max=0

dynamic.2.name=net_send
dynamic.2.source=vnstat --json -i eth0 | jq '.interfaces[0].traffic.total.tx'
dynamic.2.postprocess=$1*1024*1024*-1
dynamic.2.rrd=DERIVE
dynamic.2.min=0