RPi-Monitor/RPi-Monitor #79: Landing page change breaks reverse proxy usage

harbulot, 2015-02-09:
Commit 330ba902f4e1009e08e8e2416d7e092886cd42e4 ("Set status as default landing page", currently in the devel branch) does not work behind a reverse proxy.

Let's assume an Apache Httpd reverse proxy configured like this on a RPi at `192.168.0.15`:

        ProxyPass /rpimonitor/ http://127.0.0.1:8888/rpimonitor/ retry=15
        ProxyPassReverse /rpimonitor/ http://127.0.0.1:8888/rpimonitor/

Without this commit, `http://192.168.0.15/rpimonitor/` quite happily redirects to `http://192.168.0.15/rpimonitor/status.html` (and the fact there's a prefix doesn't seem to cause RPi-Monitor any problem).

With this commit `http://192.168.0.15/rpimonitor/` sends a redirection to `http://rpi-hostname:8888/rpimonitor/`, which will fail if `rpi-hostname` is only known to the RPi or if port 8888 is blocked for the client machine. (Of course, showing the direct URL also breaks the principle of the reverse proxy to some extent.)

I realise this is work in progress on the devel branch, but it might be worth taking this into consideration for further changes.

(A possible workaround, if this change was permanent, would be to implement the redirection from `/rpimonitor/` to `/rpimonitor/status.html` in the front-end web server.)

XavierBerger, 2015-02-23:
Thanks for this feedback. This will allow me to have a better quality for the next release. I'll test this befor publishing.

XavierBerger, 2015-02-26:
Fixed. Redirection is no more used. Status page is sent instead of landing page.

closed 2015-02-26 by XavierBerger