RPi-Monitor/RPi-Monitor #82: Crashing on raspberry pi 2

mikeluyten, 2015-02-25:
After about an hour of so of running perfectly, rpimonitord crashes.

I ran the process manually using `sudo /usr/bin/rpimonitord` and I got the following after 25min:

> $ sudo /usr/bin/rpimonitord
> Web server not started because of error: Address already in use
> rpimonitord stopped because respawning too fast.

XavierBerger, 2015-02-25:
If you did some customization, remove them and redo the test.
On my side, I've an uptime of "39 days 12 hours 37 minutes 17 seconds" and I didn't experience any issue.

closed 2015-03-24 by XavierBerger

pjao, 2015-04-28:
Hi there, i've today the same problem also with raspberry pi 2!

Some times it runs for hours, others some minutes before crash with the same message:
Web server not started because of error: Address already in use
rpimonitord stopped because respawning too fast.

Any new ideias?

mikeluyten, 2015-04-28:
I did a full removal, including confit files and re-installed and all worked again for me

pjao, 2015-04-29:
Thanks! It seams that it solved the problem!

It looks that there was a problem with the database, because i'me using the same config files.