RPi-Monitor/RPi-Monitor #86: Rounding temperatures to two digits

TobHaeg, 2015-03-10:
hello in the file 
RPi-Monitor/rpimonitor/template/temperature.conf

I've Changed the line
dynamic.12.postprocess=$1/1000
to
dynamic.12.postprocess=sprintf("%.2f", $1/1000)

so the Temperature is not shown with up to three digits after the "."
Maybe only one Digit is more than the sensor realy can read out.

so
dynamic.12.postprocess=sprintf("%.1f", $1/1000)
Should be enough

jepolch, 2015-05-07:
I tried this and my temperature sometimes shows no decimal and sometimes it shows one decimal point. Here's my line:

dynamic.12.postprocess=sprintf("%.0f",$1/1000)*1.8+32

Any idea why it doesn't always leave off the decimal point? By they way, thank you for this great program! It's very useful.

XavierBerger, 2015-05-08:
In your configuration, I see **0**f when in the proposed configuration, it is proposed to use **1**f for one digit and **2**f for two digit. You should check that.
Note: I never applied such configuration yet.

jepolch, 2015-05-08:
Yes, you're right. I first tried 1f, then thought I'd see if 0f works. It does work. But, where 1f sometimes displayed one digit and sometimes two, 0f sometimes shows zero digits and sometimes one.

XavierBerger, 2015-06-19:
This example is now part of temperature.conf template provided by default.

closed 2015-06-19 by XavierBerger