RPi-Monitor/RPi-Monitor #89: Available memory shows NaNk on Arch Linux

dragonnn, 2015-03-20:
On Arch the "Available memory" doesn't work, here is my output of the free -mk command:
              total        used        free      shared  buff/cache   available
Mem:         251348       60264       14892         204      176192      176928
Swap:        574552        1560      572992

There is no +/-buffer/cache in new procps package.

XavierBerger, 2015-03-21:
RPi-Monitor is not a "plug and play" software, it is a tool you can customize to your need. 
If, like in your situation, it doesn't display the expected value, it means that you have to update your configuration files. 
`rpimonitor -i`(Interactive Configuration Helper) should help you during your customization.
Note: Arch linux is not officially supported - I make my possible to make the job of people porting RPi-Monitor on this OS easier but I do not garanty anything.

dragonnn, 2015-03-21:
I know, just had big problem with understanding how the perl regex works, but fixed it, this works fine:
Mem:\s+\d+\s+\d+\s+\s+\d+\s+\s+\d+\s+(\d+)
BTW small offtopic :P - why rpimonitor? It works fine on my old Intel PC and is really useful. Thank you for this project.

closed 2015-03-21 by dragonnn