RPi-Monitor/RPi-Monitor #92: Upgradable packages aren't updated

Germain67, 2015-04-09:
Hello,
I've found that upgradable packages listed on the status page were never updated (the same packages are always listed, even after sudo update & upgrade).

http://i.imgur.com/FKDXVfL.png

Do you have any idea to fix this ?

Thank you in advance,
Germain

XavierBerger, 2015-04-09:
This is not a **RPi-Monitor** question but a question more related to debian based distro. I think thet the comma you are looking for is `sudo apt-get dirst-upgrade`. You can also use `aptitude` instead of `apt-get`.

closed 2015-04-09 by XavierBerger

Germain67, 2015-04-10:
Even with your command it didn't changed anything.
http://i.imgur.com/ZOazSgg.png

In fact, updates are done, 
but for a reason I ignore the old updates are still listed in rpimonitor (and the new ones aren't listed).

XavierBerger, 2015-04-10:
Your probleme is a desynchronization between RPi-Monito and the packages. I let you read why and how to fix it in the FAQ: http://rpi-experiences.blogspot.fr/p/rpi-monitor-faq.html

Germain67, 2015-04-10:
Thank you it's working now ! :)

XavierBerger, 2015-04-12:
Have a look on the Internet to understand how cron is working and how to write an init script. And you will find your solution...

woodytheledge, 2016-04-30:
I had the same issue... I ran this command and it seems to work.. 

Sudo /usr/share/rpimonitor/scripts/updatePackagesStatus.pl