RPi-Monitor/RPi-Monitor #94: Version 2.11-beta3, test and validation - help needed.

XavierBerger, 2015-06-07:
**A new model of publication**
I'm changing the way to publish **RPi-Monitor**. Until now, I was immediately publishing stable versions. From now, when version will be ready for publication, I will create a beta version and ask user to try and test this version. From my point of view, the quality of beta version will be the same as the previous release but it will let me the chance to correct or improve some features before official release.

**Why changing the model of publication**
**RPi-Monitor** has more and more functionalities and testing them become more and more complex. You are a user of **RPi-Monitor** and want to try the new feature in advance? You can then install this beta version and give your feedback on new functionalities and share your feeling about the current developments.

**Share your feedback and feeling about this version**
This issue has been created to gather your feedbacks.

Everything is working fine for you?
Share your experience or just let me know it works.

You find something not working as expected or you find a bug?
Share your experience here or open a new issue.

**When will stable version be publish?**
It's depends on you... If I have feedback saying that all the new feature are working correctly on your machine, I'll then be confident enough on the code of **RPi-Monitor** and I'll publish the next stable version soon.

Thanks for your contribution.

Xavier

dschaper, 2015-06-08:
Options popup is blank with no options to chose.

XavierBerger, 2015-06-08:
Bug description is quite small but... I did understood what you are meaning. In fact, the bug looks like that:
When you browse http://RaspberryPi/, status page is displayed correctly and option is empty.
When you browse the same page using http://RaspberryPi/status.html, status page and the option are displayed correctly.
All other options are displayed correctly.

dschaper, 2015-06-08:
Yes, the options are blank on the status screen, where the auto-reload option used to be when the page is accessed directly at http://servername:8888 and the reload option appears again when the status page is accessed at http://servername:8888/status.html. Options are okay to set default on the statistics page, and are blank for addons page.

dschaper, 2015-06-08:
A feature request. When the status screen is wide enough the gauges for CPU uptime and Temp should be to the right of the section icon instead of below. The whitespace that is there now looks a bit off. 
[image]
[image]

XavierBerger, 2015-06-09:
**Option**
Option are not always present for addons so, in next release, I'll write *"No option available for this addon"*. This is true for **Top3** and **About**.

**White space**
This is a new feature. I would then consider this as out of the scope of this beta. You can open a new issue and I'll see if I can implement it...

XavierBerger, 2015-06-11:
**Version 2.11-beta4**
I just publish version 2.11-beta4 fixing issues related to option dialog box.
I think I could try to add the label of selected graph for statistic option. I need to think about how to do this since this could also be a parameter depending on statistic shown...
Your feedback about option (and everything else) are welcome.

dschaper, 2015-06-11:
As soon as the deb is updated I'll grab it and test it out. I've opened a new issue for the whitespace reflow, do you want to discuss the options for that on the new issue?

XavierBerger, 2015-06-12:
Oups, I forgot to push... Beta4 is now in repository.
Note: You created a new issue for white space and this is the good location for comments about that new feature request.

dschaper, 2015-06-12:
I just ran an apt-get update/upgrade and rpimonitor install updated raspbian.conf but then gave the error:

    mv: cannot remove "/usr/share/rpimonitor/web/stat/": Not a directory  

I tried a purge and install to start fresh and had the same error. But the options bug is fixed.

XavierBerger, 2015-06-13:
Fixed in beta5. This didn't have any impact on installation process and was present since 2.7 and active only when you update from > 2.7 to new version

dschaper, 2015-06-13:
Okay, I did seem to have a problem restarting the service when that error occurred, the service said it was active but the webpage timed out. Probably a coincidence, the Pi that it'm testing on is up to Jessie with systemd. The fresh install fixed my dead webpage issue but I think it's a fault on my end and not the rpimonitord.

XavierBerger, 2015-06-13:
**RRD Structure Customization**
The main new feature brought by version 2.11 is the customization of RRD Structure. This allow you to configure the number of data to store, the frequency and so one. ***RPi-Monitor** is managing 5 RRD/graphs per extracted data. 
I need now some feedback about this feature. Is anyone already try it? Is the man page clear enough to let you customize the RRD? Any feedback are welcome.

dschaper, 2015-06-15:
Fresh install on RPi 2 Wheezy with beta 4 and all seems ok. I have not tested the RRD changes because I don't know of a situation that needs the adjustments. Will changing the RRD structure of an existing KPI result in a new RRD and loss of existing data? Can you provide an example using the new RRD arguments like you have for other new features?

XavierBerger, 2015-06-18:
Version beta5 is fixing install message error and adding example.customrrd.conf showing how to customize RRD structure. Is the new feature easier to understand?

dschaper, 2015-06-23:
Sorry, I didin't see the new build, I'll take a look and see what I can find out for you.

markus80, 2015-06-28:
W: Error catching https://github.com/XavierBerger/RPi-Monitor-deb/raw /master/devel/Packages HttpError404

rpimonitor.list
deb https://github.com XavierBerger/RPi-Monitor-deb/raw/master/devel/

XavierBerger, 2015-06-28:
@markus80 You did a mistake when you modified your `rpimonitor.list`. You should reinstall the stable version then follow the instruction ( http://rpi-experiences.blogspot.fr/2015/06/rpi-monitor-version-211-beta-is-ready.html ) to migrate to beta. Everything should then work.

markus80, 2015-06-28:
now it is correct:
deb https://github.com XavierBerger/RPi-Monitor-deb/raw/devel/repo/

i should know what the command 'sed -i "s/master/devel/" /etc/apt/sources.list.d/rpimonitor.list' does, if i change manually... :P ty

XavierBerger, 2015-07-25:
I publish RPi-Monitor version 2.11  beta6.
A known bug is present: When **interval** is used in configuration, a graph using constant value will not shown data correctly. The time scale will have a tick of **step** instead of **"step*interval"**. Issue is due to **empty.rrd** generation which is not taking in account **interval**

dschaper, 2015-07-25:
I seem to be having a problem with a new install from the deb file. The stat/load rrd's aren't being generated. The status display shows the graphs correctly, but when I go into statistics I get 3 popups telling me that stat/load1.rrd, load5.rrd and load15.rrd are not valid. The rrd files do not exist in the /var/lib/rpimonitor/stat directory.

XavierBerger, 2015-07-26:
I did had this issue during development and fixed it but didn't push it correctly in github :-(.
I just published Beta 7 which is fixing this issue.

dschaper, 2015-07-26:
Looks like that resolved the issue using both dpkg install for debs and also with Arch installs. I'll let you know when the PKGBUILD is up on the Arch AUR, it's almost done, just trying to figure out how to handle backing up files during upgrades so they aren't overwritten.

dschaper, 2015-08-25:
PKGBUILD for devel branch is up https://aur4.archlinux.org/packages/rpimonitor-dev-git/

markus80, 2015-09-21:
with the current version i can see 0 upgradables packages, but if i move the mouse over the magnifying glass i can see some packages - i think these are the latest upgradables ones.

x2k13, 2015-11-01:
On latest Raspbian image (Jessie), fully updated/upgraded:

Swap
Used: 0B (NaN%) Free: 0B Total: 0B

Any idea what could cause this?
All the other values show correctly.

XavierBerger, 2016-08-19:
I did remove the possibility to customize RRD structure per KPI. This feature have a lot of side effect and bug. that the reason why 2.11 is not officially published yet.
Beta version are available into a new repository hosted by http://giteduberger.fr. 
To test `v2.11-betaX` You can update you `apt` configuration with this file: [/etc/apt/sources.list.d/rpimonitor.list](https://raw.githubusercontent.com/XavierBerger/RPi-Monitor/devel/src/etc/apt/sources.list.d/rpimonitor.list) and perform installation with `apt-get`.
Closing this issue. You can open new issue if you find other bugs.

closed 2016-08-19 by XavierBerger