RPi-Monitor/RPi-Monitor #95: Feature Request: Whitespace in Status Layout

dschaper, 2015-06-10:
Current status layout in 3 panel display leaves whitespace to the left of section heading icons with the graph below the section heading:
[image]

Possibly move graphs to the left in the whitespace to compact the layout and create a better visual flow
[image]

XavierBerger, 2015-06-13:
Good news, this feature is already implemented but it may be visible or no depending on the size of the screen.
In fact, this feature may need some fine tuning. Everything is located into the file `rpimonitor.css`. 
The parameter `width` of `.Title` defines the size of left title (icon + Temperature in your example).
The parameter `width` of `.Text` defenes the size of information on right (or under): the gauge.
Finally, the parameters `@media screen and ...` define how the columns are rendered.
All these parameters have impact on how the information are displayed.

So, if you find some value that have a better render on screen, I'll be happy to integrate this on the next release.
Please describe the size of your screen and the expected result if you propose new values.

dschaper, 2015-06-13:
Okay, I'll have to refresh on my css and layout flow, but I'll take a look and report back what I find.

Thanks!

closed 2016-06-02 by dschaper