RPi-Monitor/RPi-Monitor #96: Get the disk space of an extrernal drive

GCorbel, 2015-06-19:
Bonjour,

Is it possible to monitor space of external drive. For example, I have two hard drives connected and I want to know if on of them is full. Is it possible ?

Thanks!

XavierBerger, 2015-06-19:
Yes it is possible. You should have a look to my blog I already wrote article about such features ( including example). I also add a parameter named 'visibility' to hide data from Web interface when disk is unplugged.

closed 2015-06-19 by XavierBerger

GCorbel, 2015-06-19:
Do you speak about this article :
http://rpi-experiences.blogspot.ca/2013/06/rpi-monitor-advance-usage-and.html.
I checked it and the way described seems to be very different in the last
version. Am I right ?

Thanks for your beautiful job.

XavierBerger, 2015-06-19:
Visibility is desribed here: http://rpi-experiences.blogspot.fr/2014/10/rpi-monitor-version-29-is-available.html
Disk configuration is the example used in configuration helper article: http://rpi-experiences.blogspot.fr/2014/03/rpi-monitor-interactive-configuration.html