RPi-Monitor/RPi-Monitor #97: Whitespace after regexp=(.*) causes error

bismosa, 2015-06-22:
If i have a whitespace after "regexp=(.*)" in config file i get no values. Find this error is very difficult.

Thank you for this great tool!

Regards
Sascha

XavierBerger, 2015-06-22:
A regular expression is something where everything has a meaning. Adding a space at the end have an explicit mean and will require a space to match. You should find info on Internet to understand the detail of regex.
Note: `rpimonotord -i` will help you define the correct regex

closed 2015-06-22 by XavierBerger

bismosa, 2015-06-23:
Hello,

thank you for your response. 
Ok, this seems also a mistake by me. 
I never had a whitespace at the end of a regex. Mostly a $ or a *. I had generated the (.*) and made a mistake by copying. But i have thought that the ")" is the end of the regex...and all other whitespaces do not care.