RPi-Monitor/RPi-Monitor #98: Name in Configuration to long?

bismosa, 2015-06-22:
Hello,

i have created a copy of the sdcard.conf. If i use 
static.30.name=sdcard_root_total_openelec
no sdcard_root_total_openelec.rrd file is been generated.

I do not know if it is on _ or at length...or what else.

Regards,
Sascha

XavierBerger, 2015-06-22:
There is no limitation in name length.
Static is not generating rrd. Static values can be drawn in graphs but will not have RRD associated. The curve is draw by you client by a calculation base on empty.rrd.
I would advise to read execute `man rpimonitord.conf` carefully and start from examples given (doing one change at a time) and see the impact of your changes.

closed 2015-06-22 by XavierBerger

bismosa, 2015-06-23:
Hello,
thanks for your response.
Sorry, i have copied the wrong value. I mean 
dynamic.30.name=sdcard_root_used_openelec

This generates no rrd-file. But perhaps I had also made another mistake

Regards,
Sascha