RPi-Monitor/RPi-Monitor #100: Hang on boot

thardbarger, 2015-06-28:
Hello,

I have an Arduino Micro connected via USB to my Rasberry Pi that causes RPi-Monitor to hang on boot.  If I unplug or tap the reset button on the Arduino the RPi-Monitor process will start and the RPi will finish booting.  Once booted RPi-Monitor works as expected.

Thanks,

Tom

XavierBerger, 2015-06-28:
With default configuration there is no reason for `rpimonitord` to hang.
Are you sure that **RPI-Monitor** hangs and not something else in the boot process?
If you are sure, did you customize the configuration file and are you sure that hangs soesn't comes front this script?
`rpimonitord` is managing a timeout and cancel a command if answser take too long time so, in theory, it should not hang...

thardbarger, 2015-06-28:
Hey,

Thanks for getting back to me.  I'm 99% sure it's rpimonitord, hooked RPi up to a monitor and watched boot process then uninstalled RPi-Monitor and the problem went away.  I've not done any customization and installed via your source and apt-get  How long is the timeout?  I also have a USB wireless adapter in my RPi that works fine with RPi-Monitor.  I'll toy around with it this evening.

Thanks again,

Tom

XavierBerger, 2015-06-28:
By default the timeout is set to 5 sec but can be customized with the parameter `daemon.timeout` in `daemon.conf`

thardbarger, 2015-07-04:
Hello,

Having another go.  Does rpimonitord start a log file?  I've reinstalled and it still hangs.

Thanks,

Tom

XavierBerger, 2015-07-09:
Version 2.11 (version beta for the moment) will introduce log file. With this version, it is possible to change the log level in live by sending `SIGUSR1` and `SIGUSR2` signals.

closed 2015-07-25 by XavierBerger