RPi-Monitor/RPi-Monitor #101: daemon.datastore has no effect on statistic web page if /var/lib/rpimonitor/ still exists!

MiGeppert, 2015-07-09:
I use a ramdisk for datastorage and make a backup every night (0:00).

The new data is stored on ramdisk (date changes), but on the statistics webpage, data ends on 0:00 the day before (backup time).

After renaming /var/lib/rpimonitor/ (to e.g. /var/lib/rpimonitor_backup) everything works fine!

XavierBerger, 2015-07-09:
Could you share your script to let me the chance to understand what you are doing and what the issue could be.
Note: The size of data (RRD) generated by **RPi-Monitor** never grow so it is not necessary to delete them

MiGeppert, 2015-07-09:
Hello again

As I want to reduce the write cycles to the sdcard I copied the /var/lib/rpimonitor to /run/shm and changed/added
daemon.datastore=/run/shm/rpimonitor (in daemon.conf).
Every night (0:00) there is a "cp -R /run/shm/rpimonitor/* /var/lib/rpimonitor" (at rpimonitor start it's vice versa).
But when I open the statistics page, date line ends at 0:00 -> data is used from backup (/var/lib), not from /run/shm.
Workaround: rename /var/lib/rpimonitor to /var/lib/rpimonitor_backup (so /var/lib/rpimonitor doesn't exist anymore) and timeline on statistics page is up to date.

To reproduce the bug:
copy (NOT move/rename) /var/lib/rpimonitor to other_location, add deamon.datastore=other_location, restart rpimonitor, wait 1 hour, open statistics page - timeline will end 1 hour ago until you delete/rename /var/lib/rpimonitor. (Please verify :-))

Do you still need my script? (If yes: which one?) I changed daemon.conf and /etc/init.d/rpimonitor (added copy procedure in start + stop)

Best regards
MiGeppert

BTW: ° works! I use mc/mcedit for editing the files, they were saved in ISO-8859, thus ° were replaced by the dummy character <?>. '& deg;' didn't work, was printed as '& deg;' :-(

XavierBerger, 2015-07-27:
I understood that at startup, you copy data from you backup to the ram disk where data of **RPi-Monitor** are stored. Do you preserve the right and ownership of these files? 
The issue you have may come from `empty.rrd` which is dynamically generated by the web server on demand. The web server is run as standard user, if you change the ownership of `empty.rrd`, it will not be possible for the web server to update this file and you graphs will be stuck.

MiGeppert, 2015-07-27:
Hello again
Actually I have no access to my raspberry so I can't check the rights.
But as it works when I rename the "original" directory name, I think it's not a right-problem.

Did you make the test i suggested in my last comment? "To reproduce the bug:"

Best regards
MiGeppert

XavierBerger, 2015-07-27:
You said `it works when I rename the "original" directory name` and this is exactly why I think about a right problem. When you rename and restart **RPi-Monitor** the directory structure is recreated with the correct right and ownership.
Ididn't try to reproduce the bug (not enough time ...) but your description miss some details: for example what is the user doing the copy. That why I was asking for your script.

MiGeppert, 2015-07-27:
Maybe, but the "original" directory isn't restored (after renaming).
At the moment, the original directory is named "rpimonitor_backup" and data is written/read to/from /run/shm/rpimonitor - and it works fine!

XavierBerger, 2016-08-19:
I just tried you script and I confirm, `empty.rrd` is owned by `root`:
```
ls -al /run/shm/rpimonitor/stat/empty.rrd
-rw-r--r-- 1 root root 188K août  19 15:13 empty.rrd
```

It should be owned by the user running the web service usually `uid=1000`:
```
ls -nal /var/lib/rpimonitor/stat/empty.rrd
-rw-r--r-- 1 1000 1000 191880 août  19 15:16 /var/lib/rpimonitor/stat/empty.rrd
```

You should add a `chown` in you script just after the copy to fix your issue.

closed 2016-08-19 by XavierBerger

MiGeppert, 2016-08-19:
Hello again
Ok, you are right! It works if I make a chmod 777 on empty.rrd in /run/shm/rpimonitord/stat.
But there is one thing I don't understand:Why does it work if I delete /var/lib/rpimonitord? (without the chmod?)
Anyway - the workaround is okay for me and my script uses the rpimonitord.backup directory, so no need to change it.
Best regards
MiGeppert

XavierBerger, 2016-08-19:
It works when you delete your directory because the main process of **RPi-Monitor** (run by root) create the file `empty.rrd` with the correct ownership. This correct ownership allow the web server process (owned by user 1000) to update this file.