RPi-Monitor/RPi-Monitor #109: Remove bananian-temp-daemon.sh from distribution

ThomasKaiser, 2015-07-29:
You included some stuff I wrote (quick&dirty) a while ago for Banana Pi with RPi-Monitor 2.10.

Thanks for that but unfortunately bananian-temp-daemon.sh has serious flaws regarding safe temp file handling. I rewrote the whole stuff yesterday and let now testers have a look: http://forum.armbian.com/index.php/topic/155-testers-wanted-sunxi-adjustments-for-rpi-monitor/#entry843

It would be good to replace the Banana Pi centric stuff (especially the quick&dirty written daemon) with my newer generic sunxi stuff since with the adjustments I made RPi-Monitor is able to serve all A10/A20 boards and not just Banana Pi:

http://linux-sunxi.org/Category:A10_Boards
http://linux-sunxi.org/Category:A20_Boards

XavierBerger, 2015-07-29:
Hello, 
No problem to update the bananian configuration file. The easiest and safest solution would be a pull request, can you update your conf and propose such a pull request? I'll integrate it in v2.11.
Rgds,
Xavier

ThomasKaiser, 2015-07-29:
I'll do that, just wait for some feedback. BTW: Thanks a lot for your work. Highly appreciated!

closed 2015-07-29 by ThomasKaiser