RPi-Monitor/RPi-Monitor #112: Cpu voltage on RPi2 (osmc) shows undefined

simple-elf, 2015-08-05:
Command "vcgencmd measure_volts core" from terminal showing voltage (volt=1.2000V). Regex working ok (i'm also trying "(*.)"). But on web page it still undefined.

Just new RPi and fresh install of osmc and upgrade al packages and kernel to "Linux 4.1.3-2-osmc armv7l".
Nothing changed in rpimonitor config

[image]

simple-elf, 2015-08-05:
Ha. Found quick fix.

Changed:
dynamic.2.source=vcgencmd measure_volts core
To:
dynamic.2.source=/opt/vc/bin/vcgencmd measure_volts core

XavierBerger, 2015-08-08:
This is normal. A FAQ is explaining th reason. I close this issue.

closed 2015-08-08 by XavierBerger

ahmedelakkad, 2022-01-30:
> 

Where did you change it please?

Feriman22, 2023-01-05:
@ahmedelakkad in the /etc/rpimonitor/template/cpu.conf file.