RPi-Monitor/RPi-Monitor #113: CPU JustGageBar Max Value dependent on card. - Enhancement

dschaper, 2015-08-05:
Would it be possible to use the value from `nproc` to compute the max value for the GageBars? It's hard-coded to the magic number of 3, and with a RPi 2 quadcore a system load of 4 is possible if all the cores are maxed out. `nproc` shows the value of 4 for the four cores.

XavierBerger, 2015-08-08:
It is possible (in version 2.11 beta) to used the result of a parameter as source of data extraction of another parameter. So yes it should be possible to use such a parameter to automatically configure justgage. Pull request is welcome to make it a default...

dschaper, 2015-08-08:
Should it just grab the number of processors (cores) from nproc or should it be a multiplier? (The default was to use 3 as the max on a single core Pi, but I'm not sure if 12 is a good number for the 4 core pi 2.) If we go by just nproc values then the default for single cores would be 1.

XavierBerger, 2015-08-08:
Normally, the load average should not exceed the number of core. So it could be possible to add 1 to the value extracted by using the post process formula.

dschaper, 2015-08-23:
Okay, I have the following:
```
static.9.name=max_proc
static.9.source=nproc
static.9.regexp=(.*)
static.9.postprocess=$1 + 1
```
and
`web.status.1.content.1.line.1=JustGageBar("Load", "1min", 0, data.load1, data.max_proc, 100, 80)+"`

I'm not sure how to add to both cpu.conf and cpu_arch.conf though, can they both reference static.9?

XavierBerger, 2015-08-23:
It should be added on both cpu configuration. The id (9) should be unique per file and will not generate conflict if it is declared in different file. The name should be globally unique if not, I guess that the latest read value will be the one used.

dschaper, 2015-08-23:
Okay, I will add to both cpu templates. Since I can't think of a situation where both cpu.conf and cpu_arch.conf would be used at the same time I don't think the non-globally unique name would be an issue..

XavierBerger, 2015-08-23:
Pull request are now merged on devel branch

closed 2015-08-23 by XavierBerger