RPi-Monitor/RPi-Monitor #115: statistic of /boot partition not displayed

GaetanCambier, 2015-08-19:
i haven't statistic for boot partition : 
/boot Used: NaNkB (NaN%) Free: NaNkB Total: NaNkB

detailled statistic also impacted with this error : 
File stat/sdcard_boot_total.rrd is nor a valid RRD file

could you check ?

GaetanCambier, 2015-08-19:
i have found the problem
i use ubuntu 15.04 and the first partition is not mounted at /boot but /boot/firmware
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
i have corrected in the config file and all is ok now :)

closed 2015-08-19 by GaetanCambier

eddiem5, 2025-01-08:
How weird - I have this problem 9 yrs later and don't understand what was updated in the config file as per the post above..