RPi-Monitor/RPi-Monitor #118: Upgradable package icon remains after upgrade

dschaper, 2015-08-23:
After getting a package update notification, updating packages and running both the `updatePackagesStatus.pl` and manually running `systemctl start rpimonitord-upgradable` the package count resets to 0 but the icon remains and hovering over the icon lists the old package(s) to be upgraded. 

[image]

XavierBerger, 2016-08-19:
I don't use **Arch Linux**. I only integrate the pull request proposed for this distro.
It looks that the script `updatePackagesStatus.pl` updating the file `/var/lib/rpimonitor/updatestatus.txt` have a bug. Pull request fixing this issue is welcome (in `devel` branch of course)

XavierBerger, 2016-08-19:
Unless I initially thought, this issue wasn't located into `updatePackagesStatus.pl` but in javascript.
I managed to reproduce it and fix it.

XavierBerger, 2016-08-28:
It looks that this bug is still present (but now in perl part...)

Version 2.11.
- Install a brand new `Raspbian Jessie Lite`
- Install **RPi-Monitor**
- Browse `rpi:8888` and activate auto-refresh option
- Execute `sudo /etc/init.d/rpimonitor` update
- execute `sudo apt-get upgrade`
- In Web interface, the number of package to upgrade is 0 but list of package is still displayed
- `rpi:8888/dynamic.json` show that the information has been sent by the perl script

TimJDFletcher, 2016-11-13:
The problem is that the apt-get hook is called on dpkg (install / remove etc) actions not apt-get update. The correct hook is this: 

# Update rpimonitor status
```
APT::Update::Post-Invoke {
  "echo 'Update rpimonitor Packages Status' && /usr/share/rpimonitor/scripts/updatePackagesStatus.pl";
};
```

TimJDFletcher, 2016-11-20:
Just submitted a pull request to resolve this

PR #173 (2017-08-16): Correct apt config hook for package updates
Package hook should be on apt-get update not dpkg changes
Closes #118
commit 20a8a2e1 (2017-08-16): Correct apt config hook for package updates

Package hook should be on apt-get update not dpkg changes
Closes #118
commit 375a4497 (2017-08-26): Fix #118
closed 2017-08-26 by XavierBerger