RPi-Monitor/RPi-Monitor #125: Manage visibility of parameter per line

XavierBerger, 2015-09-09:
From a comment on rpiexperiences.blogspot.fr:

Hello Xavier,

I like your application, it's running well on my different Raspberrys. But I'm missing one thing: The possibility to change the visibility of single lines in the "Status View". I can fade out a content-group, but I'm not able to do this for a line. 

Example: (this is working fine for the group) 
web.status.4.content.1.name=Visibility 
web.status.4.content.1.icon=version.png 
web.status.4.content.1.visibility=data.visibility 
web.status.4.content.1.line.1='Visibility: 

Control row visibility' Example: (this is not working for the line) web.status.4.content.2.name=Visibility2 
web.status.4.content.2.icon=version.png 
web.status.4.content.2.line.1.visibility=data.visibility 
web.status.4.content.2.line.1='Visibility: Diese Zeile verschwindet'

Is there an other way the achieve this gaol? 

Best regards

Thomas

XavierBerger, 2015-10-28:
"Content lines" are piece of javascript evaluated by your browser when the page is displayed.
So I guess (but never tried), that you could reach this goal with a line like this:
if ( condition ) then 
   print this 
else 
  print that (that could be nothing...)

I don't know the exact synthax but if you look at Badges and Label, you you find something close.

Once you will achieve your goal, you can post the method here and I'll add your trick in the man page.

amilino, 2019-04-09:
This could be a solution `typeof data.storage1_used === "undefined" ? "" : "<b>/storage1</b> Used: <b>"+KMG(data.storage1_used,'M')+"</b> (<b>"+Percent(data.sto..."`