RPi-Monitor/RPi-Monitor #131: Should create path to .rdd files on startup.

angerman, 2015-11-01:
I'm storing the data in `/var/run`, which is a `tmpfs`. It looks like rpimon doesn't create the initial folder structure as needed if it's not there.  My perl is a little rusty, so I wasn't able to figure it our right away.

XavierBerger, 2015-11-01:
The easiest way to solve your issue would be to add `mkdir` into the startup script `/etc/init.d/rpimonitor/`. I'll keep this issue open and see how to implement it when I'll work on the code again.

XavierBerger, 2016-08-19:
Fixed in version 2.11 - Feedback is welcome.

**Note**: The new repository is hosted by http://giteduberger.fr. 
To test `v2.11-betaX` You can update you `apt` configuration with this file: [/etc/apt/sources.list.d/rpimonitor.list](https://raw.githubusercontent.com/XavierBerger/RPi-Monitor/devel/src/etc/apt/sources.list.d/rpimonitor.list) and perform installation with `apt-get`.

closed 2016-08-22 by XavierBerger