RPi-Monitor/RPi-Monitor #138: Shellinabox does not work

andreaskoelsch, 2016-01-07:
Hi,
Ich was using an old version of RPi-Monitor until today. Everything worked fine there. Now, after the update to 2.10, Shellinabox does not work anymore in rpimonitor. It still works when I connect to it via https://127.0.0.1:4200
What could have gone wrong? I activated the addon and can see the tab "Shellinabox" in the Webinterface. However, when I click it, does not show the iframe.
Cheers

andreaskoelsch, 2016-01-11:
Does anyone else have the same issue? Do I have to do more configuration to get it working?

andreaskoelsch, 2016-01-11:
I found out, what is wrong: neither shellinabox nor rpimonitor listen for ipv6 connections, but rpimonitor wants to open a connection to http://[::1]:8889 instead of http://127.0.0.1:8889. I'll try to disable ipv6 and see if it works.

andreaskoelsch, 2016-01-11:
Still doesn't work, but the errors thrown by nginx are gone now. The problem persists, but now I do not even have an error log...

andreaskoelsch, 2016-01-11:
I gathered some more information and now know what the problem is. I am using a reverseproxy with nginx as proposed here: http://rpi-experiences.blogspot.fr/2013/05/rpi-monitor-security-and-authentication.html
The new version of rpimonitor has two problems with that:
1) in /usr/bin/rpimonitord it uses https://$host:4200/
Since we drop all traffic on port 4200 from other interfaces than 'lo', a remote host can't connect to https://$host:4200/ but still connecting to https://$host/shellinabox/ works fine

2) with the reverse proxy none of the addons works, because the URL looks like https://$host/rpimonitor/addons.html
I am not entirely sure, but I assume this is translated to http://localhost:8889/rpimonitor/addons.html instead of http://localhost:8889/addons.html

Therefore some feature is missing in the new version and we can decide: Either have
- an SSL secured, password protected rpimonitor
- rpimonitor with working addons

Would be cool if we could have both.

Cheers

XavierBerger, 2016-08-19:
Fixed in version 2.11 - Feedback is welcome.

**Note**: The new repository is hosted by http://giteduberger.fr. 
To test `v2.11-betaX` You can update you `apt` configuration with this file: [/etc/apt/sources.list.d/rpimonitor.list](https://raw.githubusercontent.com/XavierBerger/RPi-Monitor/devel/src/etc/apt/sources.list.d/rpimonitor.list) and perform installation with `apt-get`.

closed 2016-08-21 by andreaskoelsch