RPi-Monitor/RPi-Monitor #140: OSMC 2015.11-1 "can't find variable"

srutz26, 2016-01-16:
After a fresh install of OSMC 2015.1-1, and after fallowing the install instructions posted I get this when I visit the OSMC RPi-Monitor page:
[image]
Any idea's/fixes?

srutz26, 2016-01-16:
I'd also like to mention that I performed the same RPi-Monitor installation my Raspberry Pi 2 Model B running Raspbian Jessie and all is fine and works great. The device running OSMC is also a Raspberry Pi 2 Model B just for reference.

XavierBerger, 2016-08-19:
It looks like installation didn't work properly. 
The javascript `rpimonitor.utils.js` is gathering all the function marked as missing.

closed 2016-08-19 by XavierBerger