RPi-Monitor/RPi-Monitor #141: customize RPi-Monitor

basti122303, 2016-01-22:
hello,
I try to customize my rpi monitor.

Based on cpu.conf dynamic.3. I have add the following to network.conf
(dynamic.13 is a 2nd Ethernet Interface and dynamic.14 check my Internet connection) 

dynamic.15.name=datause1,datause2,datause3,datause4
dynamic.15.source=php /var/lib/rpimonitor/foo.php
dynamic.15.regexp=^(\S+)\s(\S+)\s(\S+)\s(\S+)
dynamic.15.postprocess=
dynamic.15.rrd=GAUGE

web.status.1.content.8.line.4="Usage: <b>" + data.datause1 + "</b> DataVolume <b>" + data.datause2 + "</b> Start <b>" + data.datause3 + "</b> Stop <b>" + data.datause4 + "</b>"

The script I run return the following (at the end of the line there is a \n):
5.37GB 6GB 01-01-2016 20-01-2016

At the Status page I see only:
Usage: undefined DataVolume: undefined Start: undefined Stop: undefined

How can I Debug?
Whats wrong? Thank your for any help.
Best Regards

basti122303, 2016-01-22:
Update:
I have delete the \n from php output.
Sometimes it works, but the last value is still undefined.
I don't know why.

XavierBerger, 2016-08-19:
I think I understand you issue. 
`RPi-Monitor` did process all the lines given by your source until the latest.
As latest was empty, it  didn't manage to extract data.
Issue is fixed in future release.

closed 2016-08-22 by XavierBerger