RPi-Monitor/RPi-Monitor #145: apt-get update results in size error for files in this repo every time.

Kezxo, 2016-02-11:
Maybe this is nothing but just want to share to see if it is some issue on my end or something that is happening for everyone.
This is what I get at the end of each ```sudo apt-get update```:
```
Reading package lists... Done

W: Size of file /var/lib/apt/lists/partial/github.com_XavierBerger_RPi-Monitor-deb_raw_master_repo_Release.gpg is not what the server reported 473 152
W: Size of file /var/lib/apt/lists/partial/github.com_XavierBerger_RPi-Monitor-deb_raw_master_repo_Release is not what the server reported 1428 148
W: Size of file /var/lib/apt/lists/partial/github.com_XavierBerger_RPi-Monitor-deb_raw_master_repo_Packages.gz is not what the server reported 442 152
```

floviolleau, 2016-02-12:
Please see [](https://github.com/XavierBerger/RPi-Monitor-deb/issues/6)

Kezxo, 2016-02-12:
At first I thought I was blind because I remembered searching before posting the issue but this is a different but similar repository. Sorry and thanks.

closed 2016-02-12 by Kezxo