RPi-Monitor/RPi-Monitor #147: Graphs don't show negative values

andreaskoelsch, 2016-02-27:
Hi,
I would like to log the temperature and it dropped below 0 these days. Unfortunately the graphs don't show values below 0.

```
dynamic.15.name=weather_temp
dynamic.15.source=/usr/bin/temperature_outside
dynamic.15.regexp=(.*)
dynamic.15.postprocess=
dynamic.15.rrd=GAUGE

web.statistics.1.content.8.graph.4=weather_temp
web.statistics.1.content.8.ds_graph_options.weather_temp.label=Outside temperature (°C)
```

You can easily reproduce this by creating a file /usr/bin/temperature_outside
```
#!/bin/bash
echo "-10"
```

andreaskoelsch, 2016-02-29:
I'm sorry, it's not the negative values. Can you tell me what else could be wrong here?

andreaskoelsch, 2016-02-29:
It seems to be working, now that I replaced "weather_temp" by "weather"... odd.

XavierBerger, 2016-08-19:
The process extracting the data has slightly change in latest version.
Alert example is using the same technique as yours and is working fine.

closed 2016-08-19 by XavierBerger