RPi-Monitor/RPi-Monitor #153: rpimonitord always appears to run as UID 1000

srcshelton, 2016-05-20:
... even if `daemon.user=<x>` and `daemon.group=<x>` are set to valid names or UID/GID numbers in `/etc/rpimonitord.conf`:

```
$ getent passwd nobody
nobody:x:65534:65534:nobody:/var/empty:/bin/false
$ getent group nogroup
nogroup:x:65533:
$ grep -E 'daemon\.(user|group)=' /etc/rpimonitord.conf
daemon.user=nobody
daemon.group=nogroup
$ ps -ef | grep -i rpi
root     27227     1  0 May19 ?        00:00:00 /usr/bin/perl /usr/sbin/rpimonitord ...
stuart   27247 27227  0 May19 ?        00:00:18 /usr/bin/perl /usr/sbin/rpimonitord ...
root     27248 27227  1 May19 ?        00:30:11 /usr/bin/perl /usr/sbin/rpimonitord ...
$ id
uid=1000(stuart) gid=100(users)
```

XavierBerger, 2016-06-07:
**RPi-Monitor** is no more using `/etc/rpimonitor.conf` but [/etc/rpimonitor/daemon.conf](https://github.com/XavierBerger/RPi-Monitor/blob/master/rpimonitor/daemon.conf) and [/etc/rpimonitor/data.conf](https://github.com/XavierBerger/RPi-Monitor/blob/master/rpimonitor/template/raspbian.conf) (symlink to `template/raspbian.conf`)
Can you check that you are using the last version of **RPi-Monitor** and you update the correct files.
Check the manuals if you need details (`man rpimonitord` and `man rpimonitord.conf`)

XavierBerger, 2016-08-12:
Tests are done and the feature is working when the configuration is inside /etc/rpimonitor/daemon.conf

closed 2016-08-12 by XavierBerger

srcshelton, 2016-08-14:
Sorry - should have been more specific.

I'm running RPi-Monitor as:

```
/usr/sbin/rpimonitord -b /var/run/rpimonitord.pid -c /etc/rpimonitord.conf -c /etc/rpimonitord.conf.d/data.conf
```

... and the settings (shown above) in `/etc/rpimonitord.conf` are not being adhered to.  Possibly a bug with processing alternate config files passed with `-c`?

This is with the 2.10 release version, incidentally...