RPi-Monitor/RPi-Monitor #157: Perl 5.24: Experimental push on scalar is now forbidden

therealbigear, 2016-06-07:
After upgrading to Perl 5.24, I get the following Error:
`Experimental push on scalar is now forbidden at /usr/bin/rpimonitord line 178, near "$1) "
BEGIN not safe after errors--compilation aborted at /usr/bin/rpimonitord line 308.`

VNovotna, 2016-06-08:
Seconding this. I tried to fix it like [here](https://github.com/hitode909/App-PerlPackage2PlantUMLClassDiagram/commit/bf48c554cc5d06e7079daa088a25973c531d5857), but it's more complicated than that and I don't know Perl.

Shelvak, 2016-08-22:
I don't remember which version of perl I was using. Any idea what version is working well?

dorepi, 2016-08-23:
I have the same probleme with archlinux.

XavierBerger, 2016-08-23:
I don't have a PC running Perl 5.24. A pull request is welcome.
I'm about to publish v2.11 in the coming days so pull request on master branch is acceptable.

dorepi, 2016-08-23:
I believe I have found the solution (It's the first time I try to read perl script) but as I'm not used to github, I'd rather to give you the solution before trying to pull request on master branch.
We have to replace the line 184 of /usr/bin/rpimonitord, 
 `/^include=(.*)/ and -f $1 and push($this->{'daemon'}->{'confFiles'}, $1) and next;`
by
 `/^include=(.*)/ and -f $1 and push(@{$this->{'daemon'}->{'confFiles'}}, $1) and next;`
I hope it's the good way to solve the problem.
Thanks for your awesome program

XavierBerger, 2016-08-23:
Pull request merged. Thanks for your contribution.

closed 2016-08-23 by XavierBerger