RPi-Monitor/RPi-Monitor #158: apt-get update Version 2.10

BadzeGit, 2016-06-23:
Hello,
for the version RPi-Monitor: Version 2.10 is available (http://rpi-experiences.blogspot.fr/2015/01/rpi-monitor-version-210-is-available.html)
When i write  
> sudo apt-key adv --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F    => OK
> sudo wget http://goo.gl/rsel0F -O /etc/apt/sources.list.d/rpimonitor.list   => OK
> sudo apt-get udpate

i have any errors

W: Impossible de récupérer https://github.com/XavierBerger/RPi-Monitor-deb/raw/master/repo/Packages  Mauvaise ligne d'en-tête
E: Le téléchargement de quelques fichiers d'index a échoué, ils ont été ignorés, ou les anciens ont été utilisés à la place.

Thank you

BadzeGit, 2016-06-23:
oups, it s a old version of january, but how to install a stable version 2.10 for test the beta version 2.11 with this new process ? (http://rpi-experiences.blogspot.fr/2015/06/rpi-monitor-version-211-beta-is-ready.html )

XavierBerger, 2016-06-23:
In you bug description some information are missing: On which computer do you install it? Which OS?
Anyway, the repository I proposed is base on github and github wasn't designed for that...I used also an old technic to generate ded which look not really compatible with latest distribution.
So I advise to perform installation manually by downloading the deb [here](https://github.com/XavierBerger/RPi-Monitor-deb/tree/master/repo) and use the command `dpkg -i` to perform the installation. This should work.

BadzeGit, 2016-06-23:
sorry, i have Linux raspberrypi 4.4.13-v7 with Rasbian Jessie Lite.
but your application your application date of 2015 and is incompatible.

> sudo dpkg -i rpimonitor_2.11-beta7_all.deb 
dpkg-deb : erreur : `rpimonitor_2.11-beta7_all.deb » n'est pas une archive de format Debian
dpkg: erreur de traitement de l'archive rpimonitor_2.11-beta7_all.deb (--install) :
 le sous-processus dpkg-deb --control a retourné une erreur de sortie d'état 2
Des erreurs ont été rencontrées pendant l'exécution :
 rpimonitor_2.11-beta7_all.deb

sorry for inconvenience

XavierBerger, 2016-06-23:
I don't have such OS but I just tried on xubuntu x86_64 with  dpkg version 1.18.4 and it works like a charm... 
Are you sure about you deb package? 
Mine is a valid `Debian Binary Package (format 2.0)` according to `file` 
```
$ file rpimonitor_2.11-beta7_all.deb
rpimonitor_2.11-beta7_all.deb: Debian binary package (format 2.0)
```

BadzeGit, 2016-06-23:
$ file rpimonitor_2.11-beta7_all.deb
rpimonitor_2.11-beta7_all.deb: HTML document, UTF-8 Unicode text, with very long lines

Debian `dpkg' package management program version 1.17.27 (armhf).

it is very strange

XavierBerger, 2016-06-23:
You did download a web page (HTML document) instead of the debian package. Try to open it with `vi`, you will see its content...
You should download the deb again.

BadzeGit, 2016-06-23:
exactly, thank you very much. It s good.
the package works !!!! :)

Thank you

closed 2016-06-23 by BadzeGit