RPi-Monitor/RPi-Monitor #159: Update issues rpi 3

theking86, 2016-07-04:
Hi,

when i want to update my rpi 3 i have this problem : 

W: Impossible de récupérer https://github.com/XavierBerger/RPi-Monitor-deb/raw/master/repo/Packages  Mauvaise ligne d'en-tête

E: Le téléchargement de quelques fichiers d'index a échoué, ils ont été ignorés, ou les anciens ont été utilisés à la place.

Can you tell me why ? Thanks :)

XavierBerger, 2016-07-04:
I don't have rpi3 so I can test it. 
This issue is similar to #158, I let you check it.

theking86, 2016-07-05:
Hi,

thanks for your response. it's exactly the same problem. On the other hand I managed to install it directly but I always have this problem of update.
Thanks.

XavierBerger, 2016-08-19:
Fixed in version 2.11
The new repository is hosted by http://giteduberger.fr. 
To test `v2.11-betaX` You can update you `apt` configuration with this file: [/etc/apt/sources.list.d/rpimonitor.list](https://raw.githubusercontent.com/XavierBerger/RPi-Monitor/devel/src/etc/apt/sources.list.d/rpimonitor.list) and perform installation with `apt-get`.

closed 2016-08-19 by XavierBerger