RPi-Monitor/RPi-Monitor #160: rpimonitord always appears to run as UID 1000 when started with custom config file

srcshelton, 2016-08-14:
(Sorry, can't re-open existing Issues)

Please see [Issue 153](153) -

When invoked as:

```
/usr/sbin/rpimonitord -b /var/run/rpimonitord.pid -c /etc/rpimonitord.conf -c /etc/rpimonitord.conf.d/data.conf
```

... rpimonitord v2.10 seems to ignore the `daemon.{user,group}` directives in `/etc/rpimonitord.conf`.

XavierBerger, 2016-08-16:
In RPi-Monitor 2.10, the configuration files you define are used in addition with the configuration file provided (/etc/rpimonitor/*.conf). 
In version 2.11, the file you provide overwrite the configuration provided by default.

Can you check that you don't have the file I provide still present and configured?

XavierBerger, 2016-08-16:
I foud the bug. I did invert user and group so when you use a group/user which as a name different front the user/group (and not existing) it used the default value: 1000

closed 2016-08-16 by XavierBerger