RPi-Monitor/RPi-Monitor #161: interval ignored

KrawallKurt, 2016-08-21:
I have the following in one of the included conf files:
```
dynamic.13.name=hdd_temp
dynamic.13.source=hddtemp
dynamic.13.interval=90
dynamic.13.regexp=(.*)
dynamic.13.postprocess=
dynamic.13.rrd=GAUGE
```

However, when I run rpimonitord with increased log level I get 
`Monitor::ParseCommand Parsing command 'hddtemp 2>/dev/null |'`
every 10 seconds. I need this command to happen every 900 seconds, because otherwise my hdd won't go to standby.

XavierBerger, 2016-08-21:
This feature have been introduce in early beta of version 2.11 but I had a lot of side effect so I decided to remove it (at least for this release).

To solve your issue, the solution would be to write a script extracting the data from your hdd and writing it in `/var/run/hdd_temp.data` (/var/run is normally a ramdisk) and configure `cron` to execute it every 5 minutes and then configure **RPi-Monitor** to read this file (it will do it every 10 sec as usual).

I will then close this issue with the label "wontfix"

closed 2016-08-21 by XavierBerger

XavierBerger, 2016-08-22:
Good news. I did reintegrate this feature in v2.11.
Interval was introduced with RRD customization but can work without this... so it is back.

KrawallKurt, 2016-08-22:
Nevermind, I am now getting the data via cronjob as you suggested