RPi-Monitor/RPi-Monitor #162: Web UI behaves badly after long periods without use (under Safari?)

srcshelton, 2016-08-23:
I coming back to a RPi-Monitor front-page after an extended period of inactivity, there will be an in-page white dialog box saying 'Cannot find gauge <name>' - which is not cleared with a reload: instead each dialog (and I get three - possibly because I'm showing three temperature gauges?) has to be clicked-through first.

Is there some call-back here being used here which could perhaps benefit from an inactivity timeout?

Next time it happens I'll take and attach a screenshot...

srcshelton, 2016-08-23:
[image]

IIRC, usually the RPi Monitor web UI is still visible in the background.  The numerical element of the gauge name does, I believe, vary. In this case I got sequential errors for 'gauge3279', 3278, and 3277 before the main page re-loaded successfully.

XavierBerger, 2016-08-23:
I did notice this bug few days ago and good news, it will be fixed in 2.11...

closed 2016-08-23 by XavierBerger

srcshelton, 2016-08-23:
Hurrah!  Is there a commit you could point me to so that I can backport the fix?

XavierBerger, 2016-08-23:
This is in commit 86524eb00fe3f1e8c1ab29c7769c6b4e853c1f5f.
**Note**: Version 2.11-r2 is available into the new repository [/etc/apt/sources.list.d/rpimonitor.list](https://raw.githubusercontent.com/XavierBerger/RPi-Monitor/devel/src/etc/apt/sources.list.d/rpimonitor.list). This should be the next official release.

srcshelton, 2016-08-23:
Ah, didn't notice you'd tagged 2.11 - thanks, I'll update the Gentoo ebuild!

XavierBerger, 2016-08-23:
You will see that the architecture or the repository has changed and the new Makefile which is simplifying the installation. I used it for Yocto.