RPi-Monitor/RPi-Monitor #165: Enhancement: Allow selection of single- or multi-column mode from Options dialog

srcshelton, 2016-08-24:
Prior to v2.11, RPi-Monitor displayed defined items in a single-column format.

Now, the default is a multi-column display... although in my case, the two items I care about most happen to be either side of the split, and so can no longer be viewed at the same time without scrolling.

Rather than require that the item ordering be changed, it would be a nice enhancement to provide a UI option to choose between single-column display and multi-column display.

XavierBerger, 2016-08-25:
I'll keep this idea as enhancement but right now, a solution exists, if you want to see only one column reducing the with of you browser...

srcshelton, 2016-08-25:
I'd argue that that's not really a solution... ;)