RPi-Monitor/RPi-Monitor #167: Bug: Web interface unresponsive after extended period in v2.11

srcshelton, 2016-08-24:
I have two different Raspberry Pis running RPi-Monitor: one ARMv6 Model B+, one ARMv7 Model 2.  I upgraded both to v2.11 at the same time, and confirmed that both were working correctly.

I've now come back ~18.5 hours later, and both had the status drop-down stating in red that they were unable to obtain data from the server.  On reloading the Status page for each instance, both hung and then timed-out without providing any data.

I checked both machines, and RPi-Monitor hadn't crashed - all three processes were still present.

On one machine, the Model 2, I restarted RPi-Monitor and all was instantly well.  In addition, there is no gap in the statistics, so the RRDs were still being updated.

On the Model B, I straced all three processes and could indeed see the data-extraction thread periodically waking with a burst of sustained activity.

However, If I strace the process which 'netstat' continues to show is still listening on the specified port (80 in this case but 8888 on the Model 2), then I get only:

```
Process 28638 attached
23:18:55 read(5,
```

... and this has been running since the above timestamp - which is now about 20 minutes ago.  I could successfully telnet to port 80 (either remotely or locally) and manually enter a valid HTTP request, but the strace of the listening process failed to respond at all, and the request received no response and eventually timed-out.

Given that this has happened on two independent hosts, it sounds as if there's some issue in the web-server component which only manifests after (relatively) extended run-periods.

To confirm, v2.10 and prior have always worked very well - this appears to be a new problem in v2.11.

I'll keep this frozen RPi-Monitor instance around for a bit, in case there's any additional debugging you'd like me to perform.

XavierBerger, 2016-08-25:
This is strange. I'm running **RPi-Monitor** v2.11 (afa404d372940fe06005f518e0321867fdea064d) since about a year on more than 300 set-top-box. We are using **RPi-Monitor** interface time to time for support and they have always respond to our requests.
The change between this version and the official release are wery small:

- changing debug traces
- Sending empty.rrd when the client request an unknown rrd.

Can you let me know if it occurs again? Increasing the log level and providing logs may help.

srcshelton, 2016-08-26:
The instance which I left running hasn't recovered (and the still-running strace, invoked as 'strace -vvfFtTs 128 -p <pid>') hasn't output a single additional line.

The instance which I restarted has now locked-up again, and is now running for just shy of 18 hours... although I'm not sure how long it's been blocking for.

I've turned the log-level up, but have seen nothing written to the specified logfile beyond the changing in log levels due to USR<x> signals.

I'll restart this process and capture stderr also...

XavierBerger, 2016-08-28:
I started an endurance test on Debian Jessie Lite (freshly installed) with **RPi-Monitor** installed from official repository with the default configuration. After "_1 day 05 hours 23 minutes 55 seconds_" the web interface is still responsive an continue to work. 
I'll continue this endurance as long as I can...

srcshelton, 2016-08-30:
Hmm - could you confirm what version of Perl and any relevant modules (HTTP::Daemon and HTTP::Status?) you have?

I've setup monit to alert and restart rpimonitord if it stops responding with 200 to HTTP requests - and since then the problem hasn't reoccurred.  This could be coincidence, or it could be that the service only locks-up if it isn't used for a period... ?

I have Perl-5.22.2 and HTTP::Daemon-6.10.0, HTTP::Message 6.110.0, and HTTP::Date-6.20.0.

XavierBerger, 2016-08-31:
On my raspberry pi, I have:
 - Perl v5.20.2, HTTP::Daemon 6.01-1, HTTP::Message 6.06-1 and HTTP::Date 6.02-1
Uptime is "3 days 20 hours 21 minutes 57 seconds" and interface is still responsive.

On the SetTopBox I mentioned upper, we was using 
  - Perl 5.14.3,  HTTP::Daemon 6.01, HTTP::Message 6.10 and HTTP::Date 6.02

XavierBerger, 2016-09-16:
Uptime: 16 days 09 hours 35 minutes 55 seconds ... closing issue.

closed 2016-09-16 by XavierBerger