RPi-Monitor/RPi-Monitor #169: Allow debug output to be written to logfile (if specified) and allow '$verbose' value to be read from daemon.conf

srcshelton, 2016-08-25:
If a logfile is enabled, then debug output should go to the log rather than to STDERR, and also it should be possible to specify the starting log-level in daemon.conf.

jnachtigall, 2018-02-27:
Yes, I have an issue with alert's `cancelcommand` not being raised. No idea why, there's not much in the log file. Being able to run rpimonitor with increase logging would help here :)

XavierBerger, 2018-02-27:
It is possible to increase log level during runtime by sending signal with kill (open logfile to see how).
I'm working in the doc of **RPi-Monitor** (ref: branch feature/docs and [RPi-Monitor-docs](https://xavierberger.github.io/RPi-Monitor-docs/)). I plan to release a new version when the docs will be ready. This new version will include some new features. This issue is in my to do list.
@jnachtigall Can you share your configuration file to let me the chance to check why it doesn't work?

XavierBerger, 2018-03-02:
develop branch: Loglevel can be set in `daemon.conf`. When logfile is define in command line output is written to it (instead of writing on stderr).

closed 2018-03-02 by XavierBerger

thygizmo, 2021-01-22:
Hi!
how exactly can I set the log-level to the highest possible value from the daemon.conf?
daemon.loglevel=5 ?, v? "vvvvv"?
Thanks!,
Giz

PS: Having a log of issues with the rpiMonitor stop responding and no log messages in the log except one saying:
[2021/01/22-08:23:53] RPi-Monitor has been started
                      Log level can be increased or decreased during runtime
                      by sending signal USR1 or USR2 with the command killall -USRx rpimonitord

dineiar, 2021-04-14:
Hi @thygizmo,

While I never worked on the RPi-Monitor code, I started using it recently and I'm debugging to find out why the CPU temperature monitoring is not working. So I came across the same issue: what is the highest possible loglevel?

Checking the code, `daemon.loglevel` is an integer value and I didn't found any log with loglevel bigger than **4**, so you are probably fine with `daemon.loglevel=5` to get all log messages.