RPi-Monitor/RPi-Monitor #172: Serverstatus is not refreshed

telemaxx, 2016-11-15:
With Version 2.7 "Serverstatus Label" was a new feature.
This nice feature only refreshes the label once, at Startup.
eg if i stop the service, the label keeps green.

The command is eg:
netstat -nlt

i want to test with
rpimonitor -i
then
<ENTER>
netstat -nlt
<ENTER>
<ENTER>

nothing happend.

What do i wrong?

Thanks for continue this project,
Thomas

dventurino, 2017-04-03:
Similar issue here:

If a server goes down the labels do not change to red until rpimonitor is restarted.
But if a server was down and goes up, then the label correctly becomes green.

serramat, 2017-04-03:
hi everyone,
same problem described by @dventurino even for me :-(

telemaxx, 2017-05-05:
i solved that problem, i use a cronjob to get the status
check_services.sh

```
#!/bin/sh
/bin/netstat -nlt > /mnt/RAMDisk/netstat.txt
cat /mnt/RAMDisk/netstat.txt|/bin/grep 8200|/usr/bin/wc -l > /mnt/RAMDisk/s_dlna.txt
cat /mnt/RAMDisk/netstat.txt|/bin/grep 8077|/usr/bin/wc -l > /mnt/RAMDisk/s_motion.txt
cat /mnt/RAMDisk/netstat.txt|/bin/grep 8080|/usr/bin/wc -l > /mnt/RAMDisk/s_logview.txt
cat /mnt/RAMDisk/netstat.txt|/bin/grep 8888|/usr/bin/wc -l > /mnt/RAMDisk/s_btsync.txt
cat /mnt/RAMDisk/netstat.txt|/bin/grep  139|/usr/bin/wc -l > /mnt/RAMDisk/s_smb.txt

```

My Ramdisk:

> /mnt/RAMDISK 

/etc/rpimonitor/template/services.conf:

```
dynamic.1.name=dlna
dynamic.1.source=/mnt/RAMDisk/s_dlna.txt

dynamic.2.name=motion
dynamic.2.source=/mnt/RAMDisk/s_motion.txt

dynamic.3.name=Logview
dynamic.3.source=/mnt/RAMDisk/s_logview.txt

dynamic.4.name=BTSync
dynamic.4.source=/mnt/RAMDisk/s_btsync.txt

dynamic.5.name=smb
dynamic.5.source=/mnt/RAMDisk/s_smb.txt

web.status.1.content.1.name=Services
web.status.1.content.1.icon=daemons.png
web.status.1.content.1.line.1=" "+Label(data.dlna,"==1","dlna","success")+Label(data.dlna,"==0","dlna","danger")+" "+Label(data.motion,"==1","motion","succes
s")+Label(data.motion,"==0","motion","danger")+" "+Label(data.BTSync,"==1","BTSync","success")+Label(data.BTSync,"==0","BTSync","danger")+" "+Label(data.smb,
"==2","SAMBA","success")+Label(data.smb,"==0","SAMBA","danger")

```

crontab entry:
`*/1 * * * * /home/pi/scripts/check_services.sh&`

Manuel5cc, 2017-05-06:
Any news about this? Still not working.

serramat, 2017-05-21:
I've found a solution a little more easy and I take ssh port for example.
The previous version of /etc/rpimonitor/template/services.conf are:
```
dynamic.1.name=ssh
dynamic.1.source=netstat -nlt
dynamic.1.regexp=tcp .*:(22).*LISTEN

web.status.1.content.1.name=Servers
web.status.1.content.1.icon=daemons.png
web.status.1.content.1.line.1="<b>ssh</b> : "+Label(data.ssh,"==22","OK","success")+Label(data.ssh,"!=22","KO","danger")
```
In my solution became like this
```
dynamic.1.name=ssh
dynamic.1.source=netstat -nlt | grep "tcp .*:22.*LISTEN" | wc -l
dynamic.1.regexp=(\d)

web.status.1.content.1.name=Servers
web.status.1.content.1.icon=daemons.png
web.status.1.content.1.line.1="<b>ssh</b> : "+Label(data.ssh,"==1","OK","success")+Label(data.ssh,"!=1","KO","danger")
```

What do you think about this @XavierBerger ?

dventurino, 2017-05-23:
I tried @serramat 's approach and it looks good to me.

I changed just the last line a bit:
`web.status.1.content.1.line.1="<b>ssh</b> : "+Label(data.ssh,"==1","OK","success")+Label(data.ssh,"==0","KO","danger")`

XavierBerger, 2017-08-26:
I found the issue of Interactive mode.
Line 1172: replace `$source =~ /^(.*)/;` by `$source =~ /^(\S+)/;`
This will be fixed in v2.12.

closed 2017-08-26 by XavierBerger

dventurino, 2017-08-28:
I still encounter this issue with v2.12-r0 with:

```
dynamic.1.name=ssh
dynamic.1.source=netstat -nlt
dynamic.1.regexp=tcp .*:(22).*LISTEN

web.status.1.content.1.name=Servers
web.status.1.content.1.icon=daemons.png
web.status.1.content.1.line.1="<b>ssh</b> : "+Label(data.ssh,"==22","OK","success")+Label(data.ssh,"!=22","KO","danger")
```

It updates from up to down but not from down to up.

XavierBerger, 2017-08-28:
Solution: 
dynamic.1.default=0

dventurino, 2017-08-29:
It worked ;) maybe it should be added to the services.conf template ?