RPi-Monitor/RPi-Monitor #178: Statistics.html incorrect time-frame

alogoc, 2017-03-06:
I am having an issue with the statistics page, it doesn't display the graphs on the current time-frame. For some reason I have to scroll infinite times to reach the current date's graphs, even if I use the 10s (24 hour total) display graphs.

NTP is configured and displays proper current date.
There is enough space on the disk for the stats, user has proper write permissions to write.

Not sure what else could I check?

XavierBerger, 2017-08-26:
I don't really understand the issue.
Data are written every 10sec. Does the time between the issue and now resolve the problem?

closed 2017-08-26 by XavierBerger

alogoc, 2017-09-18:
Seems the issue has been resolved after the latest update. Still now sure what was wrong but but now the graphs are being displayed properly :)