RPi-Monitor/RPi-Monitor #191: modules show null-info on debian 9 stretch

hexxa6, 2017-08-19:
Raspberry Pi 3 model 3 rev. 1.2 
Processor: ARMv7 Processor rev 4 (v7l)
Distribution: Raspbian GNU/Linux 9 (stretch) < fresh setup!
Kernel version: Linux 4.9.41-v7+ armv7l
Firmware: #1023

**my installation routine**

sudo apt-get update && sudo apt-get upgrade
sudo wget http://goo.gl/vewCLL -O /etc/apt/sources.list.d/rpimonitor.list
sudo apt-get install dirmngr
sudo apt-key adv --recv-keys --keyserver keyserver.ubuntu.com 2C0D3C0F 
sudo apt-get update
sudo apt-get install rpimonitor
sudo apt-get update && sudo apt-get upgrade
sudo /etc/init.d/rpimonitor update
sudo shutdown -r now 

**open with pc**
http://192.168.178.230:8888
check version: 2.11-r5 by Xavier Berger

Modules not working:
Memory 
Swap
Network
Storage

[image]
[image]

Firesphere, 2017-08-21:
When I run `free -mk` it responds with
> free: Multiple unit options doesn't make sense.

fanabordo, 2017-08-21:
same problem for me with same raspbian version.

martignoni, 2017-08-24:
Indeed `free` usage did change in Stretch, so the memory module doesn't work anymore on Stretch.

One possible solution:

1. Edit file `/etc/rpimonitor/template/memory.conf`
1. Locate line reading `dynamic.15.source=/usr/bin/free -mk`
1. Replace this line and the following (in my version: lines 21 and 22) with
```bash
dynamic.15.source=/proc/meminfo
dynamic.15.regexp=MemAvailable:\s+(\d+)
```
1. Save file and restart rpi-monitor service: `sudo systemctl restart rpimonitor`

martignoni, 2017-08-24:
Note: I can't reproduce the network problem, so don't have a solution.

Firesphere, 2017-08-24:
Almost:
> Used: -295.64GB (-61094.12%) Available: 296.13GB Total: 495.53MB
I removed one line too many.

I don't have the network problem on this pi, I'll see if I can get one to reproduce the issue though. I've got a bunch of pi's here :)

Thanks! I hope this change will make it into an official `stretch` release?

XavierBerger, 2017-08-26:
I did work today to fix issues related to the new Raspbian (Stretch).
The problem listed here should be fixed on next release.
Keep in mind that **RPi-Monitor** is a tool designed to be customized to your need and it will never cover all the situation without customization. The default configuration files proposed are installed in a directory named **template**. These templates are showing example of what can be done and should help you during the customization.
I then close this issue. Anyway, thanks for the feedbacks.

closed 2017-08-26 by XavierBerger

Firesphere, 2017-08-26:
👍 
Although I agreed they're just examples, it is kind of awkward when the examples are not working, right? ;)

XavierBerger, 2017-08-27:
@Firesphere
The default configuration is working if you use the distribution use, for development and test: the latest Radpbian available when the release have been published...
If you use something different or more recent and if you want to have working template you can propose a pull request. I will be happy to integrate it in next release. You can see in template a lot of conf file for various target which have been proposed by users.