RPi-Monitor/RPi-Monitor #194: Network statistics not available after last update

modem, 2017-08-27:
Hello,

my rpimonitor installation updated (with sudo apt-get upgrade) and now there's no network statistics, it says the following:
`To activate network monitoring, edit and customize network.conf

Help is available in man pages:

man rpimonitord or man rpimonitord.conf`
I've tested it in Jessie and Stretch with the same results.

In About->About the version shown is 2.12-r0

XavierBerger, 2017-08-27:
Due to a change on network interface name, it is not possible to create a generic configuration file which will work on every installation. If you want to activate this feature, you will need to do a customization of network.conf (as said in the message).

Keep in mind that **RPi-Monitor** is a tool designed to be customized to your need and it will never cover all the situation without customization. The default configuration files proposed are installed in a directory named template. These templates are showing examples of what can be done and should help you during the customization.

closed 2017-08-27 by XavierBerger

DarthKeizer, 2017-09-06:
@XavierBerger I updated the network.conf file and it still shows the same error message?

XavierBerger, 2017-09-06:
@DarthKeizer You have to restart RPi-Monitor. Configuration file are read once at start.

DarthKeizer, 2017-09-08:
@XavierBerger Okay, btw if I want a network monitor for both eth0 and tun0 (OpenVPN)?