RPi-Monitor/RPi-Monitor #195: Change in output of memory values

Jaxom99, 2017-09-04:
Hello Xavier, (and thanks for all your work so far),

Since my last system update, "memory" charts on my RPi3-raspbian have been very low for free and available memory.  The return of `user@rpi:~ $ free -h` is as follow
>     .........     total       used       free     shared    buffers     cached
>     Mem:          858M       823M        35M       108M        16M       161M
>     -/+ buffers/cache:       644M       214M
>     Swap:           0B         0B         0B
But the charts display ~70MB of available memory and 35MB of free memory. 

It seems to me that RPi-Monitor records the output of the first line, whereas the valuable info is now on the second line (214MB of available memory). I'm not sure what changed in the system, but I think a change is needed to keep providing the right info.

I'd be glad to help the project, even though I am no expert coder. Maybe you can give me some pointers on what/where to change, and I'll provide a PR of some form...

telemaxx, 2017-09-25:
rpimonitor uses the output of
cat /proc/meminfo

you can change the calculations easily in the config file:
/etc/rpimonitor/template/memory.conf

Jaxom99, 2017-09-25:
Thanks for this insight :) I will look into that.
It went back to normal though, but I know how to tweak it now.

closed 2017-09-25 by Jaxom99