RPi-Monitor/RPi-Monitor #199: Error uninstalling RPi-Monitor

97carmine, 2017-11-12:
I uninstalled RPi-Monitor using sudo _apt-get purge rpimonitor_ but when executing the _sudo apt-get update_ command, the terminal shows this by screen:

> Update rpimonitor Packages Status
sh: 1: /usr/share/rpimonitor/scripts/updatePackagesStatus.pl: not found
Leyendo lista de paquetes... Hecho
E: Problem executing scripts APT::Update::Post-Invoke 'echo 'Update rpimonitor Packages Status' && /usr/share/rpimonitor/scripts/updatePackagesStatus.pl'

XavierBerger, 2017-11-12:
You did install the "install_auto_package_status_update" and forgot to uninstall it
You can do it manually with the command `rm /etc/apt/apt.conf.d/99rpimonitor > /dev/null 2>&1`
I'll label this issue as bug to call  `remove_auto_package_status_update()` when package is uninstalled

97carmine, 2017-11-12:
Ok, the problem is solved, thanks

XavierBerger, 2018-02-08:
Command to remove this `99rpimonitor` file is now present in `prerm` script of **RPi-Monitor** deb package.

closed 2018-02-08 by XavierBerger

davide1995, 2020-12-10:
2020 and I need to run manually `rm /etc/apt/apt.conf.d/99rpimonitor`

JappeHallunken, 2022-03-30:
A user on reddit had the same issue, 8 days ago.
[https://www.reddit.com/r/dietpi/comments/tjzgeg/update_problems_after_removing_rpimonitor/](url)