RPi-Monitor/RPi-Monitor #200: Mail of push notification if value passes certain threshold?

jnachtigall, 2017-12-09:
This is more of a question than bug report. I looked at various tools to monitor the outside temperature using a Raspberry Pi connected to a Arduino. RPi-Monitor is the best I could find so far (I also looked at IP-Control, monitorx, Domoticz, FHEM).

I just want to connect the Raspberry via bluetooth to an Arduino which has a temperature and huminity room. The Raspberry is in the office, the Arduino in the warehouse 10m away (wall in between). Everything nicely possible with RPI-Monitor :)

Only thing I am missing: If the temperature goes below a certain degree, say 3°C, then I would like to receive an email or push notification e.g. via Pushbullet.

I've seen this is possible with monitorx or FHEM (https://wiki.fhem.de/wiki/Pushbullet, only german) or monitorx (http://www.monitorix.org/features.html), but would something be also possible/configurable with RPi-Monitor?

(I think I would also be ok with hacking something together as a shell script via cron as long as the data can be easily queried)

jnachtigall, 2017-12-09:
Looking at http://www.pratermade.com/2014/08/use-pushbullet-to-send-notifications-from-your-pi-to-your-phone/ it seems like integrating Pushbullet is pretty easy. So as long as the temperatur can be easily pulled from RPi-Monitor I guess I could get something working.

XavierBerger, 2017-12-09:
You should look to _Alert management_ feature of **RPi-Monitor** everything is explained at the end of man page explaining the format of configuration file: `man rpimonitord.conf`.
I would be interested to have a return on experience of this feature and how you implement it in configuration file.

jnachtigall, 2017-12-09:
Thanks, I'll definitely report back. (Probably, only at the start of new year, this will be a Christmas project 😄 )

Is the man page or configuration file online somewhere so I can read up on it already (without installing first). I search the repo with various keywords now but could not find anything.

XavierBerger, 2017-12-10:
The configuration file and its documentation is available here [src/etc/rpimonitor/template/raspbian.conf](https://github.com/XavierBerger/RPi-Monitor/blob/devel/src/etc/rpimonitor/template/raspbian.conf )
The manpage is generated for the comment on the top of this file.

jnachtigall, 2017-12-10:
This looks terrific, exactly what I was looking for. (You might want to add the very exact info as in the Man Page about the Alert Management to the website itself, because I think it's not described anywhere -- at least I was searching the whole web for doing this together with temperature + raspbian). Now I found something => Excited and happy :)

Closing this as there's nothing actionable. I'll report back when I added Pushbullet to this Alert Management. Thanks again!

closed 2017-12-10 by jnachtigall

jnachtigall, 2018-02-18:
> I would be interested to have a return on experience of this feature and how you implement it in configuration file.

@XavierBerger 

I implemented the system now together with my son. It was a lot of fun and RPi-Monitor played an important role in it, so I would really like to thank you - it is a really great piece of software!

What we did:

1. Have a NodeMCU microcontroller (ESP8266) to which a temperature sensor is connected. Very similar to https://rpi-experiences.blogspot.de/2013/07/rpi-monitor-use-dht11-or-dht22-humidity.html
2. This NodeMCU was placed in Room A and send its temperature via WiFi to a very minimalistic php script running on the Raspberry Pi in Room B. Just a very simple http-GET request similar to the example in https://github.com/esp8266/Arduino/blob/c8dbfb160b5c1387322eedd9499f3bbf814760a0/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino
3. The php script saves the temperature from the http GET to a local file.
4. RPi-Monitor reads the temperature from this file.

And then there's the alert script using Pushbullet. The notification is very simple: Every notified user (I have 4) needs to install the Pushbullet App. Then the raise command is a simple curl request, I wrapped in a `sendTempMail.sh` bash script. Then send the script to the user's email (the email with which the user registered at pushbullet):

```bash
curl -u $PUSHBULLETAPIKEY: https://api.pushbullet.com/v2/pushes -d email="mymail@gmail.com"   -d type=note -d title="Alert: Temperature tool low" -d body="Something in the body"
```

The overall config for the alerts looked like:

```bash
## too cold
alert.too_cold.active=data.uptime>120
alert.too_cold.trigger=data.frtemp<4
alert.too_cold.maxalertduration=900
alert.too_cold.cancelvalidation=600
alert.too_cold.resendperiod=7200
alert.too_cold.raisecommand=/home/pi/bin/sendTempMail.sh --withpush > /tmp/alert.cold
alert.too_cold.cancelcommand=/home/pi/bin/sendTempMail.sh --withpush --cancel > /tmp/alert.cold.cancel

```

In the end we used pushbullet because you can send 100 messages per month for free and that seems enough, but there are also nice alternatives like [PushOver](https://pushover.net/) which I liked better but costs 5 € per device. There's even a [cli for the Signal Messenger](https://github.com/AsamK/signal-cli), that would have been my preferred way but it requires Java and for crypto a lot of CPU so I did not use it. Then I could have set up a Signal group and send the notification there instead of individual notifications - maybe next time...

So eventually RPi-Monitor is a great fit altogether for something like this :) 

Thank you!

sebastiaanspeck, 2018-06-08:
@jnachtigall could you help me to implement Pushbullet for mine RPi-Monitor? I want to get notificiations if one of my services is KO, the temperature of the RPi rises higher than a certain threshold and if the disk usage is higher than a certain threshold.

jnachtigall, 2018-06-08:
@sebastiaanspeck Try it yourself first please. By reading the docs above and the manual. It's all documented more or less. Sorry, that I cannot help any further (just not much time these days). 

But if you run into problems I guess you can open a ticket here