RPi-Monitor/RPi-Monitor #202: statistics.html gives 'File stat/uptime.rrd is not a valid RRD archive!' after fresh installation

jnachtigall, 2018-01-28:
I installed RPI-Monitor from the Rasbian package, disabled the build-in http server and configured it via apache with
`Alias "/rpimonitor" "/usr/share/rpimonitor/web"`

It works and I can access the web interface via `http://my-raspi-ip/rpimonitor` :-)

However, when I click on `Statistics` then I get the javascript alert: `File stat/uptime.rrd is not a valid RRD archive!`.

[image]

I looked in the config if I forgot to configure something, but could not find anything. Also on the hard disk for an `stat` directory, but cannot find anything. 

The alert comes from https://github.com/XavierBerger/RPi-Monitor/blob/d096103cc82f6fed16133e037fafcadd8c88f7c7/src/usr/share/rpimonitor/web/js/rpimonitor.statistics.js#L95

Any idea what's wrong?

jnachtigall, 2018-01-28:
ok, got it working with:
`sudo ln -s /var/lib/rpimonitor/stat /usr/share/rpimonitor/web/stat`

XavierBerger, 2018-02-04:
Fixed in devel branch

closed 2018-02-04 by XavierBerger