RPi-Monitor/RPi-Monitor #205: Voltage: undefinedV (RPi v2.1.2 on OSCM Pi3)

mackonsti, 2018-02-13:
Hello and thank you for your great tool, I installed it on my Pi3 running OSMC thanks to your article.

The status web page on the CPU section shows:

`CPU frequency: 600MHz   Voltage: undefinedV`
`Scaling governor: ondemand`

Although a rather power user on MacOS X, I am not too familiar with Linux but if you'd point me to the correct file to perhaps fix this, I'd be grateful.

Seems the same may be needing for a similar display issue in Swap section on that same web page:

`Used: 0B (NaN%) Free: 0B Total: 0B`

Many thanks again.

XavierBerger, 2018-03-02:
I'm working on [documentation](https://xavierberger.github.io/RPi-Monitor-docs/index.html). I still have some review to perform but it could be a goos starting point to solve you issue.

closed 2018-03-02 by XavierBerger

mackonsti, 2018-03-02:
Merci, Xavier