RPi-Monitor/RPi-Monitor #206: addons top3 display undefined

jepiclo, 2018-02-20:
J'ai activé le plugin dans data.conf et le resultat est undefined.
A tout hasard j'ai rajouté le top3.cron en crontab,même resultat

XavierBerger, 2018-02-20:
Incomplete bug report. To use addons, ref the folowing link. 
https://rpi-experiences.blogspot.fr/2015/01/rpi-monitor-version-210-is-available.html

closed 2018-02-20 by XavierBerger

XavierBerger, 2018-02-20:
You are lucky, I'm working on the documentation and I'm just starting to review addons.

Try that:
```
  web.addons.2.name=Top3
  web.addons.2.addons=top3
```
I replaced 5 by 2. The `id` in addons is important and should start by 1 and incrementing. If you commented out the example addons, `id` should be 1.

jepiclo, 2018-02-20:
Youpiiiiii
Merci beaucoup