RPi-Monitor/RPi-Monitor #207: alert's cancelcommand is not raised

jnachtigall, 2018-02-27:
This is my configuration. For a reason I do not understand the `cancelcommand` is not raised - whereas the `raisecommand` is successfully raised/triggererd. By looking at the configuration do you see a reason for this? Maybe I got something wrong.

```
## too cold
# pi must be bootet at least 2 min
alert.too_cold.active=data.uptime>120
alert.too_cold.trigger=data.frtemp<3
# Mail and Push only after 15min
alert.too_cold.maxalertduration=900
# Cancel if 10min long less
alert.too_cold.cancelvalidation=600
# 2h later again alert if still active
alert.too_cold.resendperiod=7200
alert.too_cold.raisecommand=/home/pi/bin/sendTempMail.sh --withpush > /tmp/alert.cold
alert.too_cold.cancelcommand=/home/pi/bin/sendTempMail.sh --withpush --cancel > /tmp/alert.cold.cancel
```

XavierBerger, 2018-02-27:
I don't see obvious issue. 
You can see alert real time parameters in dynamic.json : http://raspberrypi.local:8888/dynamic.json
```json
"too_cold": {
  "lastsendalertdate": 0,
  "currentalertdate": 0,
  "alertstate": 0,
  "currentcanceldate": 0
}
```
Everything is 0 for me since I didn't start to work on this issue... but you should see something which may help you.

XavierBerger, 2018-02-28:
I checked with the example provided by default (explained here: https://xavierberger.github.io/RPi-Monitor-docs/27_configuration_templates.html#alerts) and everything looks OK: The scripts are call and files are updated. 
Can you confirm tell me if my example is working or not?
Note: I did the test on latest commit of develop branch (but I didn't touch the code of Alert in `rpimonitord`)