RPi-Monitor/RPi-Monitor #209: Update rate for services?

Woersty, 2018-03-07:
Hello,
I use the services.conf to monitor a selfmade daemon but the status doesn't change until a restart of the rpimonitor.

Is there no auto-refresh? In this case it's a bit useless, isn't it?

Best regards
Christian

XavierBerger, 2018-03-07:
This is not normal. KPIs are refreshed every `delay` seconds (default = 10).
Can you check that in your file you have default values defined
Something like that : `dynamic.1.default=0`

Woersty, 2018-03-10:
Thanks. Works.

closed 2018-03-10 by Woersty

0-kaladin, 2020-03-15:
@XavierBerger I had this same issue.  but I didn't want to define the dynamic.1.default for every dynamic.x.default i had..  since i was fine with the daemon default of 10.  However none of my defined services were refreshing when i stopped them.  When i uncommented out daemon.delay=10 and restarted rpimonitor they all refreshed as you'd expect.  So I think that default isn't actually being set.