RPi-Monitor/RPi-Monitor #211: Issues loading page

NLstn, 2018-03-18:
Hi,

I have some issues with RPI Monitor, where i need your help. I have a raspberry in my local network, on which i have installed RPI Monitor and it was all fine. But sometimes, the RPI Monitor page won't load at all. Sometimes it at least shows the title in my web browser, sometimes it loads for 5 mins+ until i close the tab and sometimes it says Page stopped working or something like that after 30 seconds. I had multiple fresh installs of raspbian stretch trying this out, every time the same result. So my questions are, do you know this problem? Did anybody ever report/solve it? And can you tell me, which log file could point me to were the issue is? Do you have some kind of debug mode so i can tail a log file and see the issue?

I really don't want to swap to another monitoring solution, because i like the information and the looks of RPI Monitor.

Thanks a lot for your help.

XavierBerger, 2018-03-18:
> do you know this problem?
> Did anybody ever report/solve it? 

No, I never remind such an issue. Personally, I used Firefox and everything goes well.

> And can you tell me, which log file could point me to were the issue is? 
> Do you have some kind of debug mode so i can tail a log file and see the issue?

You can simply stop `rpimonitord` daemon. Start `rpimonitord` manually with the option `-vvvvvv`. The number of `v` is defining the log level.

If the problem is on you browser, see what is displayed in the console.

Note: If you have perform some customization, revert them an use only default configuration file. This will help you to understand what is going on.

You can also refer to the documentation but be careful, the doc is written for next version of **RPi-Monitor** and some parameters may change or not be present. Refer to changelog (in doc) to see what differ.

pucelev, 2018-03-28:
I have the same issue with RPIMonitor opening both of my RPI 3 for the last few months.  I've tried to open it in Chrome, Firefox, Opera and Edge, all are the same.  Sometime it takes over 5 minutes to load or after 30 seconds it tells me the page stopped working.  I also tried removing and reinstalling it but nothing changed.

XavierBerger, 2018-03-28:
Sorry, but I don't have enough detail to tell something else than : It works for me... 
I use RPi-Monitor on Raspberry Pi and other embedded devices (based on Yocto) and I never experienced such issue...

closed 2018-03-28 by XavierBerger

wilbertvolkers, 2018-05-18:
Sorry to say I have the same vague issue. It seems to occur more quickly if I am using the interface. The log doesn't show a thing / seems to do things twice at a time. After the delay (daemon.delay) the issue is resolved for the moment. So if you have that at 10sec you may never notice it. I configured it to be 120 sec.

Raspbian GNU/Linux 9 (stretch) Linux 4.14.34-v7+ armv7l

Modify delay: sudo nano -l +17 /etc/rpimonitor/daemon.conf

Show logs: 
* sudo systemctl stop rpimonitor
* sudo rpimonitord -vvvvvv

Browsers: Firefox, IE, Edge

I tried to include some of the log, my edits and comments in -- CAPITALS --

* [rpimonitor-problem-notes2.txt](https://github.com/XavierBerger/RPi-Monitor/files/2016496/rpimonitor-problem-notes2.txt)

wilbertvolkers, 2018-05-20:
I took a peek in the perl code (rpimonitord) and saw there is a "Check if the server is up" part in "Monitor->Run" that is called once every "daemon.delay". I don't know why the server stops working but in my case it does. It seems to happen sometimes when requesting a page. I broke up the delay in 10s parts and check the server every 10s and the problems goes invisible. Note that the delay will now be in 10s increments.

+/- line 676 in rpimonitord:

`sudo nano -l +676 /usr/bin/rpimonitord`

My changes:

```
...

for(;;)
{
 ...

 # tempo before next process
 $configuration->{'daemon'}->{'delay'} or last;
 #sleep $configuration->{'daemon'}->{'delay'};

 my $keeplooping = 1;
 my $waited = 0;
 while($waited < $configuration->{'daemon'}->{'delay'}) {
   sleep 10;
   # Check if server is up
   if ( !$configuration->{'daemon'}->{'noserver'}) {
 	$keeplooping = kill(0,$serverpid);
 	$this->Debug(2,"Check server active $keeplooping");
 	$keeplooping or last;
   }
   $waited = $waited + 10;
 }
 $keeplooping or last;
}

...

```

trsrm, 2021-09-12:
I have the similar issue. Sometime opening the webpage gets very slow and freezes to 20-120 seconds. After it opens, page reload works OK

wilbertvolkers, 2021-09-13:
I configured RPi-Monitor to use a real webserver like Nginx:

Make a virtual directory:
```
sudo nano /etc/apache2/sites-available/raspberrypi.conf
  server
  {
    listen      80;
    server_name webserver.org;
    autoindex off;
    index index.htm;
    location /rpimonitor {
        alias /usr/share/rpimonitor/web;
        allow all;
    }
  }
```

Disable the built-in 'webserver':
```
sudo nano /etc/rpimonitor/daemon.conf
	daemon.noserver=1
```

Make rdd files available to Nginx:
```
cd /usr/share/rpimonitor/web
sudo ln -s /var/lib/rpimonitor/stat stat
```